tyhenry/CheapStepper #1: Setting 4 non default pins does not work and a 2 motor example would be useful

drp0, 2016-11-11:
On ide 1.6.12 the non default call CheapStepper stepper(4, 5, 6, 7);
results in the default pins (8, 9, 10, 11) being set.

if line 41 of the library .cpp file is altered from
CheapStepper();
to
    for (int pin = 0; pin < 4; pin++) pinMode(pins[pin], OUTPUT);
the new pins work.

The Non-blocking functions are very useful!
My example allows simultaneous control of two stepper motors with simple left, right, forward and backward functions:

````C++
/*
 * Stepper_cheap_newMoveTo3.ino
 * Controlls 2 motors at once with simple
 * forward, backward,
 * left and right functions
 * D.R.Patterson
 * 15/03/2016
 * 
 * ///////////////////////////////////////////
 * using CheapStepper Arduino library v.0.2.0
 * created by Tyler Henry, 7/2016
 * ///////////////////////////////////////////
 * 
 * This sketch illustrates the library's
 * "non-blocking" move functions -
 * i.e. you can perform moves with the stepper over time
 * while still running other code in your loop()
 * 
 * This can be useful if your Arduino is multi-tasking,
 * but be careful: if the other code in your loop()
 * slows down your Arduino, the stepper motor may
 * slow down or move with a stutter
 * 
 * //////////////////////////////////////////////////////
 */

// first, include the library :)
#include <CheapStepper.h>

// next, declare the stepper
// and connect pins 8,9,10,11 to IN1,IN2,IN3,IN4 on ULN2003 board

CheapStepper stepper1(8, 9, 10, 11);
CheapStepper stepper2(4, 5, 6, 7);  

const unsigned int steps = 4096;
const unsigned int halfRev = steps / 2;
boolean onedone = false;
boolean twodone = false;

void setup() {

// let's run the stepper at 12rpm (if using 5V power) - the default is ~16 rpm
stepper1.setRpm(12);
stepper1.setTotalSteps(steps);
stepper2.setRpm(12);
stepper2.setTotalSteps(steps);
  
/* 
the default steps is 4096 and could be omitted
is geared 63.68395:1 (measured) rather than 64:1 (advertised)
which would make the total steps 4076
for more info see: http://forum.arduino.cc/index.php?topic=71964.15
*/
  
// let's print out the RPM to make sure the setting worked
Serial.begin(115200);
  while(!Serial) yield();
Serial.print("Stepper1 RPM: "); Serial.print(stepper1.getRpm());
Serial.print(", Stepper2 RPM: "); Serial.println(stepper2.getRpm());
Serial.print("Stepper1 delay (micros): "); Serial.print(stepper1.getDelay());
Serial.print(", Stepper2 delay (micros): "); Serial.println(stepper2.getDelay());
Serial.println();
forward(360);
}

void loop() {
static unsigned int counter = 0;

// we need to call run() during loop() 
// in order to keep the stepper moving
stepper1.run();
stepper2.run();

  // if the current moves are done...
  if (!onedone)if (stepper1.getStepsLeft() == 0) onedone = true;
  if (!twodone)if (stepper2.getStepsLeft() == 0) twodone = true;
  
  if (twodone && onedone){
  counter ++;
    if(counter == 8){ // lets start again
    Serial.println("____________________________________\n");
    counter = 0;
    delay(3000);
    }
    switch (counter) {
    case 0:
      forward(360);
      break;
    case 1:
      forward(360);
      break;
    case 2:
      left(90);
      break;
    case 3:
      forward(1080);
      break;
    case 4:
      backward(360);
      break;
    case 5:
      right(270);
      break;
    case 6: 
      forward(720);
      break;
    case 7:
      left(90);
      break;
    }
  }
}
  
void forward(unsigned int mydegrees){
Serial.print("Forward "); Serial.println(mydegrees); Serial.flush();  
onedone = false;
twodone = false;
stepper1.newMoveDegrees (0, mydegrees); // n degrees from current position
stepper2.newMoveDegrees (1, mydegrees);
}

void backward(unsigned int mydegrees){
Serial.print("Backward "); Serial.println(mydegrees); Serial.flush();  
onedone = false;
twodone = false;
stepper1.newMoveDegrees (1, mydegrees); // n degrees from current position
stepper2.newMoveDegrees (0, mydegrees);
}

void left(unsigned int mydegrees){
Serial.print("Left "); Serial.println(mydegrees); Serial.flush();  
onedone = false;
twodone = false;
stepper1.newMoveDegrees (1, mydegrees); // n degrees from current position
stepper2.newMoveDegrees (1, mydegrees);
}

void right(unsigned int mydegrees){
Serial.print("Right "); Serial.println(mydegrees); Serial.flush();  
onedone = false;
twodone = false;
stepper1.newMoveDegrees (0, mydegrees); // n degrees from current position
stepper2.newMoveDegrees (0, mydegrees);
}
```

David

matonym, 2017-01-17:
I've had the same issue. This should be updated in the source. Thanks for the workaround

tyhenry, 2017-02-02:
Thanks for opening this issue, and sorry for the LOOONG delay (I was out of the country for a while).
I created a dev branch with the fix.  If possible could you try with that branch just to double check it works?  If not I will try to test soon, then will merge to master and tag as v0.2.1

tyhenry, 2017-02-02:
Also please feel free to fork and make pull requests

dclobato, 2017-03-18:
Hi. I have tested this branch and it is working fine!  I suggest that you close this issue WITHOUT changing the code and focus on issue [2](https://github.com/tyhenry/CheapStepper/issues/2) as all changes from the branch proposed here is also there.

closed 2017-03-19 by drp0