tyhenry/CheapStepper #3: Reset Loop while using WebServer on 8266

neuhoffm-sh, 2017-04-04:
I am trying to run a very basic web server (relevant code included below) and am stuck in a reset loop. The starting setup line is never reached. I am using platformio to include the library and compile the code. Ultimately, I'd like to be able to trigger two stepper motors and an LED from a website hosted on the ESP. I have tried a number of variations on how to construct the stepper object with the same result each time. Any ideas? Thanks!

```
#include <CheapStepper.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

CheapStepper stepper1 (D1,D2,D3,D4);
ESP8266WebServer server(80);

void setup() {
	Serial.begin(115200);
	Serial.println("starting setup");

	WiFi.mode(WIFI_AP);
	uint8_t mac[WL_MAC_ADDR_LENGTH];
	WiFi.softAPmacAddress(mac);
	String macID = String(mac[WL_MAC_ADDR_LENGTH - 2], HEX) +
		String(mac[WL_MAC_ADDR_LENGTH - 1], HEX);
	macID.toUpperCase();
	String AP_NameString = "IOT " + macID;

	char AP_NameChar[AP_NameString.length() + 1];
	memset(AP_NameChar, 0, AP_NameString.length() + 1);

	for (int i = 0; i<AP_NameString.length(); i++)
		AP_NameChar[i] = AP_NameString.charAt(i);

	WiFi.softAP(AP_NameChar);
	Serial.printf("Connect to Wi-Fi access point: %s\n", AP_NameChar);
	Serial.println("and open http://192.168.4.1 in your browser");

	stepper1.setRpm(12);
	stepper1.set4076StepMode();

	server.on("/motor", HTTP_POST, []() {
		uint8_t degrees = server.arg("degrees").toInt();
		stepper1.newMoveToDegree(true, degrees);
		String json = String("Received");
		server.send(200, "text/json", json);
		json = String();
	});

	server.begin();

	Serial.println("HTTP server started");

}

void loop() {
	server.handleClient();
	stepper1.run();
}
```

tyhenry, 2017-04-06:
Hmm, I'll have to break out my ESP8266 to test, but sounds like it might be a power issue if the ESP is hardware resetting (esp. if you're powering the motor from the ESP directly - you should use a separate 5v DC power supply)
Have you tested w/ a different motor library like Arduino Stepper?

neuhoffm-sh, 2017-04-06:
I was using a separate power supply for the motor and the ESP. I was able to get it working using the AccelStepper library.

tyhenry, 2017-04-06:
Cool, glad you got it working - I'll leave this issue open as a reminder to
test both libs with my ESP to see why this one is choking

maxim-kukushkin, 2020-06-19:
I've got the same with external power supply. Also, the issue occurs even if the motor is completely disconnected from anything (only ESP, usb cable and PC). So, it's seems like some software failure.

I noticed the it happens when rotation is invoked for more than 90-120 degrees (regardless by steps or by degrees).

My sketch is simple:
```
#include <CheapStepper.h>

CheapStepper stepper(D1, D2, D3, D4);

void setup() {
  Serial.begin(9600);
  stepper.setRpm(10);
}

void loop() {
  stepper.moveCW(4096); // <<<-- fails here after rotating by around 120 degrees
  delay(1000);
  stepper.moveCCW(4096);
  delay(1000);
}
```

The the stack trace is:
```
0x40106530: delayMicroseconds at /home/myuser/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_wiring.c line 205
0x40106561: __digitalWrite at /home/myuser/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_wiring_digital.c line 82
0x40202159: uart_flush at /home/myuser/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/uart.c line 656
0x402028cc: CheapStepper::seq(int) at /home/myuser/Arduino/libraries/CheapStepper/CheapStepper.cpp line 292
0x401011c4: ets_timer_arm_new at ?? line ?
0x402028fa: CheapStepper::seqCW() at /home/myuser/Arduino/libraries/CheapStepper/CheapStepper.cpp line 190 (discriminator 1)
0x4020295d: CheapStepper::step(bool) at /home/myuser/Arduino/libraries/CheapStepper/CheapStepper.cpp line 158 (discriminator 1)
0x40202dac: esp_yield at /home/myuser/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_main.cpp line 91
0x4020298c: CheapStepper::move(bool, int) at /home/myuser/Arduino/libraries/CheapStepper/CheapStepper.cpp line 48 (discriminator 2)
0x402010b2: delay at /home/myuser/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_wiring.c line 54
0x40202761: loop at /home/myuser/Arduino/stepper_test/stepper_test.ino line 19
0x40202e38: loop_wrapper at /home/myuser/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/core_esp8266_main.cpp line 125
0x401000e5: cont_wrapper at /home/myuser/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/cont.S line 81
```

Hope this helps

maxim-kukushkin, 2020-06-20:
A bit more googling suggested that the key thing here is:
```
Soft WDT reset

ctx: cont 
sp: 3ffffd20 end: 3fffffd0 offset: 01b0
```
just before the stack trace (https://github.com/esp8266/Arduino/issues/2240). Specifically "Soft WDT reset". This is ESP-specific watch dog doing reset when it doesn't get the control back in a reasonable time (and that is a huge difference from "delay()" call). Arduino doesn't have this mechanism as it doesn't have a network stack running in background and thus can do "delayMicroseconds" as much as it wants.