tyhenry/CheapStepper #5: Object Initializers issue

ondsinet, 2017-11-21:
Does the current implementation of the inizializer 
`CheapStepper (int in1, int in2, int in3, int in4);
`
work? I've found that the pins mode was not changed when declaring the stepper before the setup using it.
I'm in a hurry so I just added a begin() method that does the same as in CheapStepper() in the setup and that fixed the problem.
I hope I'm making sense and not mistaking the purpose of the library.

JGoodHub, 2017-11-25:
Just tried it with my version and i'm having the same issue, no matter what pins I declare for the motor driver it always defaults to 8 through 11 meaning only one driver board can be connected at once

tyhenry, 2018-09-20:
fixed with [v0.2.1](https://github.com/tyhenry/CheapStepper/releases/tag/v0.2.1)

closed 2018-09-20 by tyhenry