tyhenry/CheapStepper #9: Two steppers issue

AndyGFX, 2018-05-10:
Hi,

I have used 2 steppers and works only one (LEFT), 

// RIGHT
#define R_IN1 4
#define R_IN2 5
#define R_IN3 6
#define R_IN4 7

// LEFT
#define L_IN1 8
#define L_IN2 9
#define L_IN3 10
#define L_IN4 11

CheapStepper L_motor (L_IN1,L_IN2,L_IN3,L_IN4);
CheapStepper R_motor (R_IN1,R_IN2,R_IN3,R_IN4);

but with other lib like (AccelStepper + MultiStepper) works good.

Is your lib designed to using 2 steppers?

Best Regards,
AndyGFX

Eddiiie, 2018-09-15:
I think I had same issue, fixed by setting all the pins used for the steppers to output mode.
Try it?
 pinMode(R_IN1, OUTPUT); 
 pinMode(R_IN2, OUTPUT); 
 pinMode(R_IN3, OUTPUT); 
 pinMode(R_IN4, OUTPUT); 

 pinMode(L_IN1, OUTPUT); 
 pinMode(L_IN2, OUTPUT); 
 pinMode(L_IN3, OUTPUT); 
 pinMode(L_IN4, OUTPUT);

tyhenry, 2018-09-20:
fixed with [v0.2.1](https://github.com/tyhenry/CheapStepper/releases/tag/v0.2.1)

closed 2018-09-20 by tyhenry