tyhenry/CheapStepper #17: newMoveToDegree Int overflow in toStep variable

drp0, 2019-11-02:
int toStep = deg * totalSteps / 360 does not create correct value
Solved using:

```
void CheapStepper::newMoveToDegree (bool clockwise, int deg){

	// keep to 0-359 range
	if (deg >= 360) deg %= 360;
	else if (deg < 0) {
		deg %= 360; // returns negative if deg not multiple of 360
		if (deg < 0) deg += 360; // shift into 0-359 range
	}
	float f = totalSteps / 360.0; // drp
	int toStep = 0.5 + deg * f;
	newMoveTo (clockwise, toStep);
}
```

David

mrv96, 2020-01-08:
If i well understand, your code in practise rounds the float result before the int convertion. Right?

drp0, 2020-01-08:
float f = totalSteps/**360.0** forces a float result
deg *f therefore produces float
0.5 +float does indeed round the float to nearest integer
David

mrv96, 2020-01-08:
Well. I will update my fork asap and i will open a pull request.