tyhenry/CheapStepper #19: After any move the stepper remains in a high current state

drp0, 2019-11-07:
My stepper draws 200mA when moving and **370mA when the move has completed**.
It would be useful to have an idle call which sets the stepper control lines to zero:

This works:
void idle();	// drp CheapStepper.h
```
void CheapStepper::idle(){
      seq(8);      // drp CheapStepper.cpp
}
```
This usilises the default case in switch(seqNum) of  void CheapStepper::seq (int seqNum)

If the function is called on the completion of a move, the current drops to 30mA !
The next move appears un-affected.

   
```
stepper.run();
  if (stepper.getStepsLeft() == 0){
  // if the current move is done...
  stepper.idle(); // low current
```

David

mrv96, 2020-01-08:
Hi, your method idle is already present in the code:

```
`void CheapStepper::off() {
	for (int p=0; p<4; p++)
		digitalWrite(pins[p], 0);
}`
```
Consider also the fact that if you power off all coils, your motor will loose retention torque.

drp0, 2020-01-08:
Hi,
off() looks like the same idea- does it affect any other parts of the library?

I had considered the loss of torque. However in some applications battery demand in static positioning outways the need to apply torque to a stationary motor.

mrv96, 2020-01-08:
off() is already part of the master official library, it's sufficient to call it.

Since it has been already included, i suppose that it doesn't affect any other parts of the library